KiCad/kicad-templates #3: Status of kicad.pro and some other files

jsreynaud, 2018-01-15:
The following files in this repo:
 - kicad.pro
 - gost_landscape.kicad_wks
 - gost_portrait.kicad_wks
 - pagelayout_default.kicad_wks
 - pagelayout_logo.kicad_wks

Are also in main KiCad repo. Is there are supposed to be duplicate ?

Thanks,

poeschlr, 2018-01-15:
The kicad.pro was only part of the kicad-source repo in the past (It is not part of the kicad-library repo that has been used until now.)
This file held the information what libs the default project should include. (This would be the part of interest for us lib maintainers.)

Now that the sym-lib-table stuff is merged, i think the sym-lib-table located in the kicad-symbols repo should take over this role. If this file contains other default settings for new projects then the decisions lies with the main development team as to who should be responsible for it. (I would guess it is better to have it in the kicad-source repo.)

The wks files are the page layout files. (i think)
I am not sure who should be responsible for these. Including them here might make it easier for contributors to add new layout definition files. But at the same time it does remove control by the lead developers. 

@stambaughw this might require your input.

jsreynaud, 2018-01-15:
More info: I ask this because for daily build maintenance (PPA for ubuntu), I need to know what version I have to keep ...

SchrodingersGat, 2018-01-15:
@stambaughw can you weigh in here? Should these files be retained (duplicate) in this repository, or deleted?

stambaughw, 2018-01-15:
These should not be duplicated.  All that does is to make it possible
for there to be multiple variations which will cause issues.  What to do
with them depends on how kicad is packaged.  Technically the only file
that really needs to be installed with kicad is the kicad.pro file.  All
of the other files your list are optional.  If the kicad-templates repo
is going to be packaged separately as an optional package, then
kicad.pro should be remove from the kicad-templates repo and maintained
in the source repo.  Otherwise, kicad.pro can be removed from the source
repo.  I would remove the worksheet files from the source repo and
maintain them in the kicad-templates repo since they are not required
for kicad.  If no one objects to this, I can make it so.  FYI, the
kicad.pro file in the kicad-templates repo is not the same as the one in
the source repo and needs have all of the libraries removed to avoid
conflicting with the new library list.

tijuca, 2018-01-27:
The most flexible way for packages here would be to add some cmake options where the packagers can control the installation of those files e.g. -DUSE_KICAD_PRO_FILE and -DUSE_WORKSHEET_TEMPLATES

This could looks like this. As jsreynaud mentioned, for distro packagers it's important to know how to go here. I've tested the snippet and it works as I wanted this.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 1e742a1..59a43b2 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,6 +2,19 @@ project( kicad-templates )
 
 cmake_minimum_required( VERSION 2.6.1 FATAL_ERROR )
 
+#================================================
+# kicad-templates build options
+#================================================
+
+# Note: The file kicad.pro section '[eeschema/libraries]' needs to by in sync with the current installed libraries!
+option( USE_KICAD_PRO_FILE
+        "Installing a dedicated kicad.pro file with the KiCad templates, only needed if KiCad source is built without that file (default OFF)."
+        OFF )
+
+option( USE_WORKSHEET_TEMPLATES
+        "Install some worksheet files that are useful for creation of new templates (default ON)."
+        ON )
+
 #================================================
 # Locations for install targets.
 #================================================
@@ -65,16 +78,20 @@ foreach( tpl ${template_lst} )
     install( DIRECTORY ${tpl} DESTINATION ${KICAD_TEMPLATE} COMPONENT resources )
 endforeach()
 
-# The default project file.
-install( FILES kicad.pro
-    DESTINATION ${KICAD_TEMPLATE}
-    COMPONENT resources
-    )
+if ( USE_KICAD_PRO_FILE )
+    # The default project file.
+    install( FILES kicad.pro
+             DESTINATION ${KICAD_TEMPLATE}
+             COMPONENT resources
+           )
+endif()
 
-# The supplied worksheet templates.
-file( GLOB kicad_wks_files "*.kicad_wks" )
+if ( USE_WORKSHEET_TEMPLATES )
+    # The supplied worksheet templates.
+    file( GLOB kicad_wks_files "*.kicad_wks" )
 
-install( FILES ${kicad_wks_files}
-    DESTINATION ${KICAD_TEMPLATE}
-    COMPONENT resources
-    )
+    install( FILES ${kicad_wks_files}
+             DESTINATION ${KICAD_TEMPLATE}
+             COMPONENT resources
+           )
+endif()
```

poeschlr, 2018-01-27:
We will go with the route suggested by @stambaughw.
The kicad.pro file will be removed from this repo as it has no library dependent information in it and is better maintained over at the source repo.

For the worksheet templates we will keep them here. (As they will be removed from the source repo, there should not be any conflicts.)

@tijuca why would we need a modification to the make file if we go that route?

tijuca, 2018-01-27:
Because until now no information was given how upstream will work on this issue, but if this is all modified in the next days my suggestion is of course obsolete then.
I need to do something now as I go into some final packaging preparation. I've added a local patch for the package build now.

poeschlr, 2018-01-27:
Well pro file is deleted from this repo. The worksheet files removal from the source repo is a different manner.

tijuca, 2018-01-27:
> The worksheet files removal from the source repo is a different manner.

Yes, as this is need to be done in the kicad tree and doesn't affect kicad-templates directly. Due the new dependent packages, which I need to finish first, I haven't done anything in/for the KiCad source yet.

stambaughw, 2018-01-27:
I plan to get them removed from the KiCad source tree this weekend.

Cheers,

Wayne

poeschlr, 2018-01-27:
Thanks at all for your help. @jsreynaud if everything is resolved could you please close this issue?

tijuca, 2018-01-28:
The root for this report is just partially fixed by removing the kicad.pro file as this is still referenced as install target in the CMakeList.txt file and the build is failing now of course.

```
...
CMake Error at cmake_install.cmake:104 (file):
  file INSTALL cannot find "/build/kicad-templates-0~2018.01.27/kicad.pro".

Makefile:120: recipe for target 'install' failed
```
Updating the CMakeList.txt file accordingly should fix this error.

poeschlr, 2018-01-28:
Issues should be closed by the one reporting it. Not the one who thinks to fix it. (Had i added the reference, this would have already been closed. And i would not have gotten the feedback about the makefile.)

poeschlr, 2018-01-28:
Would the fix i proposed in https://github.com/KiCad/kicad-templates/pull/6 work?
(I have very limited experience with make files.)

tijuca, 2018-01-28:
> Issues should be closed by the one reporting it. Not the one who thinks to fix it.

I disagree, then you the first person I know who thinks it has to go that way. I know no single workflow with such a design where reporters of issues have to close there created reports. This is up to the people who do the administration and changes. And typically closed issues will simply be reopen if needed. But this is all up to you, if you have a bugtracker with a big amount of open cases you will see why your approach can't work in the long-term.

Back to topic.
Yes, this change would remove the installation of the kicad.pro file. But the removal of the kicad.pro file has nothing logical to do with the removal of the template, this should be split of into two commits.

poeschlr, 2018-01-28:
To clarify why we normally let the decision to close to reporters here on the lib:

In most cases the reporter knows more about the topic at hand than we library maintainers. Especially with things like this issue here.
For this reason we will not trust automatic systems to close issues. We will in such cases request the reporter to check if the issue has been resolved to their satisfaction. 

New PR with only the kicad.pro stuff: https://github.com/KiCad/kicad-templates/pull/7

jsreynaud, 2018-01-29:
For me it's ok now. 
 - *.kicad_wks files are on kicad-templates only
 - kicad.pro is on kicad source only (no more in kicad-templates)

closed 2018-01-29 by jsreynaud