KiCad/kicad-templates #8: Issues with templates

evanshultz, 2018-01-30:
From https://github.com/KiCad/kicad-templates/pull/5:

- [ ] None of the templates use hierarchical sheets.
- [ ] The following don't have the headers in a box on the schematic: `BeagleBone-Black-Cape`, `minnowboard-ls-lure`, `raspberrypi_hat`, `raspberrypi-gpio`, `raspberrypi-gpio-40pin`, `stm32f100-discovery-shield`, `ti-stellaris-boosterpack40_min`.
- [ ] The following board doesn't open in v4: `minnowboard-ls-lure`.
- [ ] `raspberrypi_hat` has a symbol called `40HAT` which is rather non-descriptive. What if another version of the Rasp Pi had a different pinout? Or another template needed a 40-pin header? Contrast this with `raspberrypi-gpio-40pin` (and the templates above) which uses a generic header.
- [ ] Similarly, `ti-stellaris-boosterpack40_min` also has specific named headers instead of generic ones.
- [ ] `raspberrypi_hat` board could also include the locations of the camera and display cutout on some info layer as a guide. It also has horrible settings for clearance and min width for the zones.
- [ ] `raspberrypi-gpio` board has a text note that P1 should be on the bottom of the board, so why isn't it placed there? For whatever reason it's on the top. `raspberrypi-gpio-40pin` does this properly.
- [ ] Only a few of the board templates indicate which side of the board should be up (or against the mother board?) when the shield is used. Or which side the headers go on. Is this kind of information important enough to be documented on each board?
- [ ] `stm32f100-discovery-shield` has a zone poured. Perhaps it's better to leave that all up to the user.

poeschlr, 2018-01-30:
None of them will open in v4. The schematics have been through the remapping process which adds the information from what lib each symbol comes from to the schematic. KiCad v4 does not understand this. So i think you can remove that particular note.

evanshultz, 2018-01-30:
All schematics open in 4.0.7 with a warning, and all boards but the one mentioned open in 4.0.7 without complaint.

poeschlr, 2018-01-31:
the minoward board has a broken footprint for the mounting holes. (It is a no longer existing footprint. That i thought represents different possible hole sizes by just not including the pad. But it has a pad that just has no active layers but still has a pin number.)

So whoever does fix this must first find out what drill size should be used and exchange these footprints for the correct footprint from our mounting hole lib.

robkam, 2018-08-17:
Having copyrighted templates!?

poeschlr, 2018-08-17:
Copyright does not mean they are copy protected. It means they are licensed under some license. In this case they are under the license found in the license.md file ;) (The copyright notice simply fulfills the -BY part of the license. We could make that more clear by adding the license name to the copyright notice.)

The templates are under CC-BY-SA (creative commons, attribution, share alike) with an exception waiving the share alike and attribution clause if you use them. (But not if you want to publish the full library)

antoniovazquezblanco, 2018-11-19:
Is it really needed at this point to have support for v4 in this template repo? Can we drop that task?

heitorPB, 2019-02-06:
I think a nice KLC to standardize the templates would be great. I opened an issue for it: https://github.com/KiCad/kicad-website/issues/352 but so far nobody started writing one :P