KiCad/kicad-templates #12: I have a macrofab kicad template design in google drive here is the link

josheeg, 2018-05-12:
I have a macrofab kicad template design in google drive here is the link
i don't understand github this is compressed in 7zip & windows 10.
kicad 4.07

https://drive.google.com/open?id=12R0N_3MhzhpNaGwH8O7papERemE6tAa-

antoniovazquezblanco, 2019-02-25:
@josheeg Thanks for the files but we cannot accept contributions unless they are made through Github at the moment. This is for various reasons but mainly one. When you contribute through Github you sign the contributor license agreement so that we know that there are no licensing issues.

I would recommend you to learn how to use Git as it is an outstanding tool for development!

Thank you for your contribution.

closed 2019-02-25 by antoniovazquezblanco

poeschlr, 2019-02-25:
Additionally: All contributions to the library must be made using kicad version 5. This is especially important for templates as version 5 introduced the symbol library table and does now include the library name in the schematic when referencing a symbol. This means a template created with version 4 does not really work for version 5 users.