KiCad/kicad-templates #14: KiCad thinks CMakeModules is a template

pointhi, 2018-06-30:
KiCad thinks CMakeModules is a template and tries to open icon and html from it

```
19:37:49: Kann Datei './kicad-templates/CMakeModules/meta/icon.png' nicht öffnen (Fehler 2: Datei oder Verzeichnis nicht gefunden)
19:37:49: Konnte das Bild aus der Datei "./Kicad/kicad-templates/CMakeModules/meta/icon.png" nicht laden.
19:37:49: Kann Datei './kicad-templates/CMakeModules/meta/info.html' nicht öffnen (Fehler 2: Datei oder Verzeichnis nicht gefunden)
```

Probably templates should get a ```.template``` suffix in the future to distinguish them from normal directories

nickoe, 2018-06-30:
I wonder if that uninstall module thing is even needed, I would have expected make uninstall to work fine with default cmake.

nickoe, 2018-10-16:
@stambaughw as you comitted the CMakeModules, do you have any thoughts on this one?

antoniovazquezblanco, 2018-11-16:
I would vote to use the `.template` suffix following what already has been done with footprints and 3D models unless any other suggestion is made by any of you.

Whenever you give the green light I will create a PR with any solution we can come up.

Thank you

antoniovazquezblanco, 2018-12-17:
@poeschlr any direction to move forward the issue?

antoniovazquezblanco, 2019-02-25:
Using the `.template` suffix would also help separate template projects and worksheet templates that, in my opinion, should not be spread in folders around this repo.

Should I proceed and rename all the project templates with the `.template` suffix?

@poeschlr @stambaughw 

Thanks!

stambaughw, 2019-03-15:
@antoniovazquezblanco, adding a .template suffix to the template paths will not have any affect until there is code in place to do this and I'm not sure this is inherently a good idea.  We force this behavior with footprint library path names and I'm not entirely happy with this solution.  I would rather let the user choose their own path names an have KiCad behave appropriately.  Unfortunately any solution for this will likely break the current behavior so figuring out a reasonable path forward is not as easy as one would like.

antoniovazquezblanco, 2019-06-13:
https://github.com/KiCad/kicad-templates/pull/21 solves the issue.

antoniovazquezblanco, 2019-08-27:
The issue has been solved by #21

closed 2019-08-27 by antoniovazquezblanco