KiCad/kicad-templates #16: Raspberry pi hat resistor issue

JoeEarlam, 2018-11-14:
The Raspberry pi hat template specifies R23 and R24 (part of the ideal diode back circuit to prevent back powering) as 22R resistors. This much, much too low - 230mA is being pulled through the transistor, through the resistor to ground! Way over the 1/16w spec for an 0402 resistor.

If you look at the official Raspberry Pi documentation, it recommends **47K** as a value for these resistors: https://github.com/raspberrypi/hats/blob/master/zvd-circuit.png

This resistor value should be changed to something more sensible - is it possible the unit (K) has been missed out of the schematic?

I can confirm this issue affects the actual hardware produced using this template - we designed a hat around it and and after a little while, burned out these resistors.

evanshultz, 2018-11-14:
First, thanks for noticing this issue and taking the time to bring it to our attention. We appreciate community involvement like this, but I'm most happy when it doesn't involve a bonehead error in the existing KiCad library repos.

Sigh. Yeah, something happened here.

The schematic even references https://github.com/raspberrypi/hats/blob/master/designguide.md#back-powering-the-pi-via-the-j8-gpio-header, which references the schematic snippet you linked above.

Note that the recommended circuit only uses one 47kohm resistor while the other should be 10kohm.

Fix submitted at https://github.com/KiCad/kicad-templates/pull/17.

JoeEarlam, 2018-11-14:
Nice quick fix, well done!

closed 2018-11-14 by JoeEarlam